LaserWeb/LaserWeb4-Binaries #2: Windows Install

zafrirron, 2017-11-03:
Installing 64 bit version (also same with x86), get blank screen!
System configuration:
Windows 10 64bit V 1709 Build 16299.19
CPU Intel i&-4770K 3.5GHz
Graphics NVIDIA GeForce GTX 760 (1920 X 1080)

jorgerobles, 2017-11-03:
Did you have a previous installed LW?

zafrirron, 2017-11-03:
nop

jorgerobles, 2017-11-03:
Hmm cannot reproduce error. I've got the same configuration (CPU and GPU) as you, and runs ok :|

zafrirron, 2017-11-03:
Ok, Thanks.
Will try to take a look at the Source Code...

jorgerobles, 2017-11-03:
It's probably not related with source code but a incompatibility with something on your computer (to be determined) as we have a bunch of Windows users that should have reported same error so far.
@cprezzi, .env works on windows?

zafrirron, 2017-11-03:
Env on Windows...

Your environment has been set up for using Node.js 8.4.0 (x64) and npm.

C:\Users\User>env
!::=::\
!C:=C:\Users\User
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\User\AppData\Roaming
ASL.LOG=Destination=file
AVR32_HOME=C:\WinAVR-20100110
AWS_DEFAULT_REGION=eu-west-1
BONJOUR_SDK_HOME=C:\Program Files\Bonjour SDK\
COMMONPROGRAMFILES=C:\Program Files (x86)\Common Files
COMMONPROGRAMFILES(X86)=C:\Program Files (x86)\Common Files
COMMONPROGRAMW6432=C:\Program Files\Common Files
COMPUTERNAME=ZAFI-MAIN
COMSPEC=C:\WINDOWS\system32\cmd.exe
GOPATH=C:\Users\User\go
GOROOT=C:\Go\
GYP_MSVS_VERSION=2015
HOMEDRIVE=C:
HOMEPATH=\
LOCALAPPDATA=C:\Users\User\AppData\Local
LOGONSERVER=\\ZAFI-MAIN
NUMBER_OF_PROCESSORS=8
ONEDRIVE=C:\Users\User\OneDrive
OS=Windows_NT
PATH=/c/Users/User/AppData/Roaming/npm:/c/Program Files/nodejs/:/c/WinAVR-20100110/bin:/usr/bin:/c/ProgramData/Oracle/Java/javapath:/c/Python27/:/c/Python27/Scripts:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/c/Program Files/Git/cmd:/c/Go/bin:/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/c/Program Files/PuTTY/:/c/Users/User/go/bin:/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit/:/c/Program Files/nodejs/:/c/Program Files (x86)/Common Files/Acronis/VirtualFile/:/c/Program Files (x86)/Common Files/Acronis/VirtualFile64/:/c/Program Files (x86)/Common Files/Acronis/SnapAPI/:%SystemRoot%/system32:%SystemRoot%:%SystemRoot%/System32/Wbem:%SYSTEMROOT%/System32/WindowsPowerShell/v1.0/:/c/Users/User/AppData/Local/Microsoft/WindowsApps:/c/Program Files (x86)/HMA! Pro VPN/bin:/c/Users/User/AppData/Local/Google/Cloud SDK/google-cloud-sdk/bin:/c/Users/User/AppData/Local/atom/bin:/c/Users/User/AppData/Roaming/npm:%USERPROFILE%/AppData/Local/Microsoft/WindowsApps:/c/Program Files (x86)/EaseUS/Todo Backup/bin/x64/
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=x86
PROCESSOR_ARCHITEW6432=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=3c03
PROGRAMDATA=C:\ProgramData
PROGRAMFILES=C:\Program Files (x86)
PROGRAMFILES(X86)=C:\Program Files (x86)
PROGRAMW6432=C:\Program Files
PROMPT=$P$G
PSMODULEPATH=;C:\Users\User\AppData\Local\Google\Cloud SDK\google-cloud-sdk\platform\PowerShell
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SYSTEMDRIVE=C:
SYSTEMROOT=C:\WINDOWS
TEMP=/tmp
TMP=/tmp
USERDOMAIN=ZAFI-MAIN
USERDOMAIN_ROAMINGPROFILE=ZAFI-MAIN
USERNAME=Zafrir
USERPROFILE=C:\Users\User
VBOX_MSI_INSTALL_PATH=C:\Program Files\Oracle\VirtualBox\
VS140COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
WINDIR=C:\WINDOWS
TERM=cygwin

C:\Users\User>

jorgerobles, 2017-11-03:
Thanks, I was referring to a .ENV file next to installation to report errors.
But since you have nodeJS installed, you could test LW on chrome, downloading github sourcecode, did you mean that on previous message?

zafrirron, 2017-11-03:
Yes I did...

zafrirron, 2017-11-03:
Works like charm on local dev server!

cprezzi, 2017-11-03:
@jorgerobles Yes, the .env file works on Windows, but it's not needed for the dev environment, because you can also change the settings directly in config.js.

closed 2018-01-11 by jorgerobles

gdenicolas, 2019-11-27:
I upgraded to 4.0.996-136 and now only have a blank screen. Maybe I need to eliminate some folders before upgrading. I am running windows 10

wtcarterofgalwayhome, 2020-11-15:
This looks like an old and common issue, but I can't find the resolution.
I just installed LW4 4.0.998-136 on my Windows 10 machine and I get a white screen on startup.

I tried uninstalling and reinstalling and tried both 32 and 64 bit... All with no luck.

Any help will be appreciated.