LaserWeb/LaserWeb4-Binaries #3: Blank screen at startup

M1K3rSz, 2017-11-08:
after re-installing latest Laserweb.dmg, blank screen at startup, deleted all files and folders on iMac and re-install again without any succes.

jorgerobles, 2017-11-08:
Same as #2, on mac.
Sorry, but currently we have no way to know what is happening inside your computer (Missed that Hogwarts class)
I personally develop and test on a Sierra OSX with GTX760 with no problem.
In order the issues to be attended, System information (at least) is needed.

M1K3rSz, 2017-11-08:
Ah Yess , offcourse ,sorry i had to post some more about my System info ;)
As told i removed lw.comm-server folder , that did the trick!

How it all happend i dont know , but thanks for the quick support

closed 2017-11-08 by M1K3rSz