LaserWeb/LaserWeb4-Binaries #6: A javascript error occurred in the main process

djjeanb, 2018-01-11:
dear sirs,
i have downloaded the newest version on a clean win 10 machine,
installed java 
installed chrome
but somehow i get this error:
A javascript error occurred in main process details see attachment
followed by a 404 message in the main screen.
what could be the problem?

[image]
[image]

jorgerobles, 2018-01-11:
You have another process occupying port 8000. Please check your OS configuration and services.

closed 2018-01-11 by jorgerobles

djjeanb, 2018-01-11:
the occupying proces is windows NT kernel and system, not something you can close

jorgerobles, 2018-01-11:
See https://github.com/LaserWeb/LaserWeb4/issues/428#issuecomment-337622494

djjeanb, 2018-01-11:
yeah tried that but window wo'nt allow blank filenames so i cant make the .env file,

jorgerobles, 2018-01-11:
Sorry, cannot help with that. You should be able to create file on console.
Btw, if your OS is NT probably  LW won't work anyway.

djjeanb, 2018-01-11:
Well it works on my other similar windows 10 machine but has the older version installed “out of the box”

I’ll keep digging for more info on this problem

Thanks anyway

djjeanb, 2018-01-11:
Got it fixed, i’ll post solution on forum

djjeanb, 2018-06-25:
you can alter the .env file to select a free port.
beware the .env is not a extention its the whole filename, just read the link i provided earlier

df8oe, 2018-06-25:
Many thanks - works perfectly. Nice peace of software! I will dig into its usage now.