LaserWeb/LaserWeb4-Binaries #7: Finishing Appveyor / Travis autocompilation

jorgerobles, 2018-01-16:
HELP NEEDED

Have no time right now to finish the autocompilation through AppVeyor/Travis of releases, nor manually compilate every change of LW.

I think the only thing it lacks is the proper api keys to distribute the releases.

jorgerobles, 2018-01-16:
@tbfleming

tbfleming, 2018-01-16:
I don't know anything about the AppVeyor/Travis setup. Actually, I don't think I've heard the name AppVeyor before. I'll post a help needed to G+.

cojarbi, 2018-01-16:
Post the procedure I'll take care of it weekly.until this is solved

jorgerobles, 2018-01-23:
Almost Done. Both appveyor for WIN and travis for MAC works. Thanks @sdesalas for your support.
Have to tweak a bit to release on the same tag (right now each service pushes individual releases)

closed 2018-01-25 by jorgerobles