LaserWeb/LaserWeb4-Binaries #10: Weird release numbers

quintesse, 2018-03-17:
Release numbers don't seem to coincide with the .exe you download? Also for the last several release the version of the downloaded .exe has stayed the same?

[image]

jorgerobles, 2018-03-17:
Known issue. Since moved to new automated build, I could not get the
versions to be consecutive with the old build process.
Pull requests are welcome. :)

<https://user-images.githubusercontent.com/778793/37560035-02f41608-2a31-11e8-99a7-187ba2b8ad2b.png>

<https://github.com/LaserWeb/LaserWeb4-Binaries/issues/10>, or mute the
thread

quintesse, 2018-03-17:
Sure, where's the build process defined?

And what _should_ be the correct numbering?

:)

jorgerobles, 2018-03-17:
It's on appveyor and travis yml on this repository, the correct version
should be 4.0.1xxx.
127 fragment corresponds to lw.comm-server version.

closed 2018-03-22 by jorgerobles