LaserWeb/LaserWeb4-Binaries #11: lw.comm-server does not start after grid set to 0

Set-Me-Up, 2018-03-21:
I messed around with the grid settings and by mistake I set the grid size to 0. The application froze immediately and I had to kill it.
Since then, i can not start the LaserWeb anymore. 
I am using Windows 10 x64 Pro.
I tried to uninstall and reinstall, cleared the installation path, cleaned registry entries, even tried an older version of LW, nothing helped. The application starts up, a short flickering with a black rectangle is shown and then the window remains white and nothing happens.
Maybe someone can give me a hint what else I could do, but i suppose this is (still) a bug.

jorgerobles, 2018-03-21:
Please check gplus community. There's a guideline to solve that issue
there. Anyway use laseeweb4 repository for this kind of issues. Binaries
repo issues are only for building issues.

jorgerobles, 2018-03-22:
Fixed on next release https://github.com/LaserWeb/LaserWeb4/commit/3f2420b20103996ecaf2c2a07548879b1dfd642c

closed 2018-03-22 by jorgerobles