LaserWeb/LaserWeb4-Binaries #13: Generate Linux distribution on this repository

jorgerobles, 2018-05-04:
@sdesalas Can you help us to make a multi .travis for run multiple build targets adding linux also?

probonopd, 2018-05-05:
You can use one `.travis.yml` file with a env matrix. Like here:

```
env:
  - ARCH=x86_64 DOCKER_IMAGE=foo/bar
  - ARCH=i686 DOCKER_IMAGE=boo/baz
```

This will make the same build run twice, but with different environment variables that you can use in your script to do different things.

jorgerobles, 2018-05-20:
Tried to do multiple oses, OSX build should work for linux too (as far as I've unified oddities), but throw broken pipes (multi-os is not very stable I guess)

matkoniecz, 2019-11-07:
> but throw broken pipes

Is it just warning in log or a serios issues breaking something?