LaserWeb/LaserWeb4-Binaries #14: 32-bit linux

loefstrand, 2018-06-03:
The laserweb-builder-v4.0.993-130-x86_64.AppImage dosen't work on 32-bit Linux.
It looks like it is only made for 64-bit.

jorgerobles, 2018-06-03:
Correct. That is why the file is suffixed x86_64

loefstrand, 2018-06-03:
Then the information in the wiki is easy to misunderstand the information "Go to https://github.com/LaserWeb/LaserWeb4-Binaries/releases and download the AppImage for x86 or x64 from linux32 or linux64 respectively".
So where can I get the LaserWeb AppImage for x86 (linux32)?

jorgerobles, 2018-06-03:
I suppose there were linux 32 builds sometime ago, but fell obsolete when
changed to travis builds.
I'm aware of this already
https://github.com/LaserWeb/LaserWeb4-Binaries/issues/13

loefstrand, 2018-06-03:
Then I can just scrap trying to use LaserWeb and continue with VisiCut instead.

loefstrand, 2018-06-04:
https://github.com/LaserWeb/LaserWeb4-Binaries/issues/14#issuecomment-394204601
Because  I have  a ASUS eeebox with SSD in perfect working condition, that I want to reuse....

goosenphil, 2018-06-24:
I'm also interested in this, since I would like to run this on a Intel Edison. It seems the build fails when installing electron with npm.

goosenphil, 2018-06-24:
I managed to get it working. All you need to do is follow the Raspberry Pi build instructions, it will throw an error when attempting to install electron, just run this command before "npm install" and it will work:
```
sudo npm install -g electron --unsafe-perm=true --allow-root
```