LaserWeb/LaserWeb4-Binaries #31: Windows 10 - error: Listening EACCES 0.0.0.0:8000

Parkrcz, 2020-01-21:
Hello,
I´v installed LaserWeb4 app on a few devices without issue, but I am struggling to make it work on my PC. I am getting error message: "error: Listening EACCES 0.0.0.0:8000", it looks like ports are in collision. I´v checked this and I found that port 8000 is already in use, but unfortunately it´s System and I´am not able to kill it. Is there a way, how to change port number for this app manually?

Thank You

Petr

cprezzi, 2020-01-28:
You can change the default port by creating a textfile named `.env` (nothing in front of the dot!) and enter the line `WEB_PORT=8001` (change 8001 to your preferred port).

closed 2020-02-11 by cprezzi

frank-pian, 2021-07-26:
I have same problem in v4.0.998-136. I put .env in installation directory. But it doesn't work.
[image]

cprezzi, 2021-12-20:
I checked it with version 4.0.999-138 and the .env file works corectly. Are you sure your .env file doesn't have any additional extention like .txt, which is just hidden by your file explorer default settings (hide known filetypes)?