LaserWeb/LaserWeb4-Binaries #33: Linux Appimage does not exist

wmahfoudh, 2020-04-25:
No links to download the Linux appimage from https://github.com/LaserWeb/LaserWeb4-Binaries/releases

yoyo42, 2020-05-01:
There are several issues about this already, going all the way back to 2018. I just read about LaserWeb and came to look for the same thing as you, but TBH it looks like all the devs lost interest in this project last year sometime and it's been dormant ever since. The Linux installer seems to have fallen away a long time before that. Shame.

jefbed, 2020-05-01:
I have this issue too, of course.  This is sad, as it is the only viable open source laser CAM. My computers are too old to run windows and I try to avoid proprietary software anyways, so I guess I'll try gcode tools in inkscape with universal gcode sender.

bianchifan, 2020-05-16:
LaserWeb-v4.0.996 AppImage is still available ;)
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/untagged-4818330b6baa8213d4a7

jefbed, 2020-05-16:
Thank you for the link!

dgoadby, 2020-05-17:
I am about to replace the controller in my ageing HPC laser and I have downloaded the sources to see what can be done. Assuming this account stays open then I will report/update here with my developments.

povi2946, 2020-06-20:
linux binaries plz.

trombik, 2020-07-15:
can someone test this? please note that the package is alpha. https://github.com/trombik/LaserWeb4-Binaries/releases/tag/v4.0.996-145-pre

jefbed, 2020-07-15:
On Devuan GNU/Linux, the main application window appears, but no UI
elements do.  It is a blank white window, though it is in the normal size
of LaserWeb.

trombik, 2020-07-16:
on my ubuntu VM, it works

[image]

there should be something missing. if you run the application from CLI, do you see anything?

```console
> ./LaserWeb-linux-amd64-v4.0.996-145-pre.AppImage
```

trombik, 2020-07-16:
note that, before launching the application on my ubuntu server, i need to install:

```console
apt-get install libgtk-3-0 libx11-xcb1 libnss3 libasound2 libdbus-glib-1-2
```

they are usually installed by other applications as dependencies, but make sure you have them.

trombik, 2020-07-16:
it worked on my VM because I have a working `lw.comm-server`. it should work without it. let me see what i am missing.

jefbed, 2020-07-16:
I ran the apt-get install, and I already had those packages.  When
launching LaserWeb, the only message is "Skip checkForUpdatesAndNotify
because application is not packed".  This is a screenshot of it:
[image: sshot.png]
And a screenshot of the executing terminal:
[image: term.png]

trombik, 2020-07-16:
i cannot investigate the issue further at the moment. the problem is the backend server invoked by the frontend does not run for some reason.

to work around it, you can download the backend and run it from the console.

```console
> wget https://github.com/trombik/lw.comm-server/releases/download/v4.0.138/lw.comm-server-linux-amd64-v4.0.138.AppImage
> chmod +x lw.comm-server-linux-amd64-v4.0.138.AppImage
> ./lw.comm-server-linux-amd64-v4.0.138.AppImage
```

after that, run the frontend.

trombik, 2020-07-16:
progress so far:

in my fork, CI is implemented in GitHub Actions so that my fix will not be evasive. however, some `scripts` in `package.json` are broken because node modules are updated. the CI builds the code for different targets in a unified CI steps. the CI produces binaries for Linux amd64, macos amd64, Windows 64 bit, and Windows 32 bit. i also noticed that macos binary is no longer available in the release page. it was said in some issues that the majority of users are Windows user. Linux users are not alone in this context.

as reported here, the binary does not work because the bundled `lw.comm-server` does not run. the server process does not log anything because the server process logs to console, but, probably because the application is GUI, the log is not available. to investigate further, I need to modify the code, which I would like to avoid at the moment because the developer does not like many changes at once.

standalone `lw.com-server` in my fork is working. if you have `nodejs` installed on your machine, `node server.js` should work. if you are familiar with packaging, you should be able to write a package. however, that will not be acceptable solution because of Windows users.

the binaries are built in an interesting way. the code in this repository provides GUI interface. the GUI application invokes `lw.conn-server`, which is in another repo, and runs an HTTP server process that serves static files in yet another repository. you need to deal with three build processes. `lw.comm-server` claims that it is independent from frontend, but the build process is tightly coupled, probably because Windows user cannot deal with web server and nodejs. 

`package.json` in all of the repositories needs love. see `npm audit` output. in my fork, many dependencies are updated while trying to fix `serialport`.

i cannot find any tests in three repositories. `npm run test` is just a wrapper of `echo no test`. upgrading dependencies will be fun.

i am working on fixing build process so that others can help fixing the build. if you are curious, see https://github.com/trombik/lw.comm-server/tree/windows-build

trombik, 2020-07-17:
@jefbed would you follow the instruction and report the result?

https://github.com/trombik/lw.comm-server/blob/windows-build/doc/Build.md

you should see something like:

```
vagrant@default-ubuntu-1804-amd64:~/lw.comm-server$ node ./server.js 
 
***************************************************************
        ---- LaserWeb Comm Server 4.0.136 ----        
***************************************************************
  Use  http://127.0.1.1:8000 to connect this server.
 
* Updates: 
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull
 
* Support: 
  If you need help / support, come over to 
  https://plus.google.com/communities/115879488566665599508
***************************************************************
```

jefbed, 2020-07-17:
It is working now following your steps.  A few clarifications based what I
had to do:  yarn is known as yarnpkg on Devuan, npm needs to be listed as a
dependency, and 'yarnpkg build' has to be run twice to achieve a successful
build under Devuan.  For now, I'll use a script to start the server before
the LaserWeb appimage.
Thank you for your work on reviving this project.

trombik, 2020-07-17:
the bug mentioned has not been fixed yet. you still need to run the server manually. this build includes major updates in its dependencies. as such, I do not know it works or not. please test it with your machine because `serialport`, the core library in `lw.comm-server` has been updated to 9.x from 6.x. `electron` is now 9.x instead of 1.6.x.

https://github.com/trombik/LaserWeb4-Binaries/releases/tag/v4.0.996-152-pre

trombik, 2020-09-28:
I tried to update `serialport` but gave up because the `lw.comserver` API does not abstract anything. the changes in serialport affects the front-end as well. I cannot fix it unless someone with knowledge of reactjs help the project. the front-end has same issues in the dependencies. the `v4.0.996-145-pre` tag in my fork is the last known working tag.