LaserWeb/LaserWeb4-Binaries #35: How to get latest as .DMG ?

ThomasPujolle, 2020-11-15:
There is something I probably don't understand, but for each latest release there is often .exe and .yaml, but no .dmg, how can I manage to get those? Thanks

albworkshop, 2020-12-16:
same issue for me, cannot find the DMG file  to download

smoki3, 2021-01-25:
Also looking for this

wndxlori, 2021-01-25:
Ok, just to save other people the time... you can run it from your browser here:

https://laserweb.github.io/LaserWeb4/

That's the only solution I found. Not sure why that's not in the documentation. It would save a lot of time and frustration.

3Dietrich, 2021-01-27:
Either you don't support "OSX" or you need to answer: **Where is the .DMG file**?
The last file I found is "untagged-.." what does it mean. It is version v4.0.996-130 - from 2018. Is this the correct DMG file ?