LaserWeb/LaserWeb4-Binaries #37: Abandoned project ?

FrankHovis, 2021-02-27:
Its over 2 years now since any activity here, has this been abandoned?

dgoadby, 2021-02-28:
It seems so. I use this so did wonder if Frank would agree to it being forked or transferred to someone who is active in the group. There are a few bugs to clear up and some improvement test I would want.

cprezzi, 2021-10-03:
The frontend devs have left the project, but I could give new devs acces to this repo, if someone is capable of taking over.

webii, 2021-12-19:
I am a fullstack dev and using this software a lot. As I am using raspberry pi zero w, LaserWeb is a perfect solution for remote working. But there are a few bugs which are really painful. I would like to fix them if I can get at least some support and code review.

cprezzi, 2021-12-20:
I am open for pull requests and will support you as good as I can. Just tell me what you need to know.

webii, 2021-12-21:
Is there any way to contact you more directly, like using private messages or discord?
First I started looking on server code. But before any serious changes I needed to do some clean up, to remove duplicated code :/ https://github.com/webii/lw.comm-server/pull/1

cprezzi, 2021-12-22:
I am still maintaining the server code. What we realy need are frontend devs. 

I saw your refactoring, which chages a lot of code according to your personal preferences. Are you willing to take over backend maintenance for the the next few years? 

You also removed the App folder, which is the hosted frontend. This is needed for server only installations (like RPi) which are used by browser over network. Did you consider all usecases (server only, server & app, app only)?

webii, 2021-12-22:
1. There is no change in logic in my MR.
2. Changes are not due to my preference, but due to the good practices, as code there were duplicated three times, and still in many places is. Because for each type of connection there was the same (almost) code. It is hardly manageable.
3. Fronted app is not needed there. It could be installed by downloading from the other repository. I will add a message in the readme for those who need hosting also frontend. As someone is working on raspberry pi this frontend is not needed as you can connect from any frontend (electron) providing raspberry pi address and port. Also providing frontend on the server side might also be misleading, because there is no way to use laser cutter for more than one person/session, also the web session cannot be shared.
4. I have removed linux binary. Those binaries could be added as artifacts when building specific release, they don't have to be part of a code repository.

cprezzi, 2021-12-22:
1. I agree.
2. I partly agree but using let instead of var (ts over vanilla) is just personal preference.
3. Many users have a pi as the server and don't want to install an app for the frontend but instead just use a webbrowser on a PC or tablet. The hosted app is also needed in the case when the electron exe is run as server and the frontend wants to be used by webbrowser. 
4. I agree.

helifino, 2022-03-26:
However, now there's no linux build in Releases. For someone trying to install to Linux desktop, but not comfortable building from sources, this is unusable.

helifino, 2022-04-03:
Please then update the Installation (Linux) page. https://laserweb.yurl.ch/documentation/installation/34-install-linux-appimage-format

cprezzi, 2022-04-04:
Thanks for that hint! I completely forgot this page.

helifino, 2022-04-24:
> The installation process for Linux is described on https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(RasPi)

This fails the no-coding-knowledge test. Had to determine and solve several dependency issues just to install, and still it does nothing. Connect via browser gives 'connecting to server' in the status pane but no connection. Linux Mint 20.3

cprezzi, 2022-04-24:
@helifino Did you get the frontend in the browser?
Did you set the correct server connection? (Go to the comms tab, click "Server Connection" edit "Server IP" to own IP:8000 and hit connect)
What was shown in the console where you started the server.js?

helifino, 2022-04-24:
Oddly, without changing anything, now it's actually connecting to itself but not to the laser:
[image]

cprezzi, 2022-04-25:
It seems like server.js doesn't have the right to open /dev/ttyUSB0. Are you sure that this is the correct interface? On Raspberry Pi, it is /dev/ttyAMA0. 
You could also try if it works when you start the server with superuser rights (sudo).

helifino, 2022-04-25:
Yes, this is not a pi.

helifino, 2022-04-26:
Point of clarification: Desktop version can communicate with machine. Comms server cannot. Launched from same user account.