LaserWeb/LaserWeb4-Binaries #38: Relaunch LaserWeb

d4mer, 2021-04-05:
Hi All,

Just wanted to see if there was any interest in taking up the project and continuing the development. I am willing to take this on if there are others also willing to jump on and dedicate  a bit of time.

Please contact me so we can chat.

wolkenjaeger, 2021-04-11:
It would be great if this project lives on. However, with a powerful AND not so pricey software like Lightburn it is questionable if this is required. For now, I did all my projects with LaserWeb.

dgoadby, 2021-04-12:
I am just upgrading my K40 to a GCODE controller so will be looking to use this software. After I have finished the upgrade I will see if I can contribute.

gfrornermichi, 2021-04-16:
I am not able to contribute as I am not a programmer, but it would be awesome as this is the best linux able software for lasers

targor, 2021-06-15:
it would be nice to relaunch this tool, maybe with .net mono...

Lightburn is indeed not a bad tool, but i am really missing some features laserweb has, for example the matrix burn and the configuration possibilities of the different layers when it gets to svg vectors, and the thread/cpu optimizations e.g. . In lightburn there are only 31 configurable laser options, which is bad when you want to maybe want to make a test matrix or have a really extensible project with many different burnings. In laserweb you could add as many items to your list as the memory can hold ^^. Also when it gets to headless server/ running on a pi .. the support people from lightburn always argument with.. you do not leave your laser alone, but there are other 
reasons why someone wants to run a program on a server.. all in all i would appreciate to have a new and open source version of laserweb.. and if it is possible for me, i would also like to participate.

cprezzi, 2021-10-03:
I am the dev of the backend (nodejs) and could supprt you, if someone takes over the frontend (react/redux).