LaserWeb/LaserWeb4-Binaries #41: white page on launch of app

jmartens2, 2021-12-06:
hi. thank you for the hard work. ive run across a problem,  i wish to request some assistance. the laserweb worked on my laptop this morning and i was trying to work thru some tutorials when the progr4am just turned the screen white. ive uninstalled and reinstalled a couple times but now if i click on shortcut it goes to white screen with laserweb in upper left hand corner. im running the current laser web from gthub. im on a windows 7 pro machine that i still use for video editing so i assume its not the machine. ive downloaded the software each time ive tried to reinstall. im kinda at loss as to why this is happening. thank you if you read this and hopely there is a solution. jmartens

cprezzi, 2021-12-20:
You could try to install an older version to solve this problem. Some times the white screen problem is caused by local firewall settings which block acces to the webserver of LaserWeb4. After installing an older version you should be asked again for the access, which you must allow.

dmartinordax, 2022-05-09:
Hello, I am having problems installing the LaserWeb-v4.0.996-130.dmg version for Mac, because I get a blank screen. I have tried to re-install previous versions and the same thing happens, I have been trying since yesterday.

I had it installed and it worked fine for me, but it must have given an error when configuring machine sizes and since then it gives me a white screen error. What other solution is there on Mac apart from reinstalling the previous version?

cprezzi, 2022-05-09:
@dmartinordax Did you try to install the new Mac version 4.1.000?
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.1.000

dmartinordax, 2022-05-10:
Yes, I try this version but didn´t work also. The version of my Mac is
12.3.1 (Monterey)
D

freakout242, 2022-11-08:
I am also having this issue. Everything was working fine until I changed the grid size, now all I get is a white screen. uninstalling and reinstalling does not work. is there anyway to revert back to the way it was?

HorribleGoat, 2023-03-06:
> I am also having this issue. Everything was working fine until I changed the grid size, now all I get is a white screen. uninstalling and reinstalling does not work. is there anyway to revert back to the way it was?

Have the same problem. Tried to adjust grid but the grid size number boxes were scaled to nearly 0 size and apparently I turned them to some program breaking value. Cant find a way to reset local settings.

cprezzi, 2023-03-06:
The settings are stored in the "Local Storage" of the embedded Chromium browser. On Windows, this data is saved under /Users/username/AppData/Roaming/lw.comm-server/Local Storage/leveldb/
Find the latest .ldb file in this folder and delete it. This should get you back to the setting before.

cprezzi, 2024-11-14:
For those who can't solve the blank white screen issue, let the app whith the white screen running and start a (chrome) browser on the url "localhost:8000". This way you can get the working frontend and try to reset the settings.