LaserWeb/LaserWeb4-Binaries #43: Apple Silicon compatibility

tokamac, 2022-08-16:
Mac computers do not run on Intel chips anymore, but _Apple Silicon_ CPUs. For two years.
It's time to upgrade your code so LaserWeb can run smoothly on M1 (and now M2) Macs!
Running through emulation thanks to Rosetta 2 is only to let developers some time to port their app as a Universal binary.

tokamac, 2022-08-16:
I now understand better this may never occur thanks to point (3) in this message: 
https://github.com/LaserWeb/LaserWeb4-Binaries/issues/37#issuecomment-999616678

tbfleming, 2022-08-16:
We Think Different by not slaving ourselves to Apple's latest whims.

tokamac, 2022-08-16:
You almost had me! For a moment I thought this troll comment was from the main author of the software. But from some guy that forked the project and hadn't touched a line of code for 6 years, I understand better ^^

Whatever, accessing the server through a browser doesn't bother me. It's fine.

closed 2022-08-16 by tokamac