LaserWeb/LaserWeb4-Binaries #46: changing grid size settings breaks program, no way to reset settings

HorribleGoat, 2023-03-06:
Im adding a separate ticket for this though in the white window startup ticket someone had had this same happen but the topic is different. 

So basically went to adjust the grid settings, the grid size number boxes were near 0 size and could not see what the number was so tried to type in new one and broke it and now the program wont start and there does not seem to be a way to reset settings. Uninstall and reinstall does nothing so local settings must be somewhere.

gustgreg94, 2023-06-20:
I had this same problem yesterday and solved changing manually the config file:
C:\Users\{{your_user}}\AppData\Roaming\lw.comm-server\Local Storage\leveldb\{{some_numbers}}.txt

This is the only text file in the folder.
To fix open the file and search for "toolGridWidth" or "toolGridHeight", and change the incorrect value.

PS: This field appears many times in this file, maybe it is a config history.

Madigans26, 2024-07-19:
The exact same thing happened to me on MAC and I have no way to fix it.
Every time I open the program it shows me a blank screen and no matter how much I delete the program and save it again, it doesn't fix it.
Something must have been recorded in some file because I don't understand that if I have deleted the program and loaded it again the problem will persist.
If anyone knows which file I should look in, please tell me.
Thank you so much.
Special thanks to the creators as the application has a very good workflow

pablogs9, 2025-02-08:
@gustgreg94 I owe you a beer, you saved me